kriswiner/ICM42688 #6: Gyro self test fails

rohithr2597, 2022-08-18:
Accelerometer self-test works fine but the gyro always fails in the self-test. I tried debugging using the library and checked everything but nothing worked. 
00> accelerometer self test
00> Ax dif : 95.1538 mg
00> Ay dif : 106.567 mg
00> Az dif : 724.304 mg
00>  should be between 50 and 1200 mg
00>  gyro self test
00>  Gx dif : 13.5269 dps
00>  Gy dif : 13.7253 dps
00>  Gz dif : 13.7405 dps
00>  should be > 60 dps 
00> ares 6.10352e-05 
00> gres 0.00762939 
00>  AX ratio: 92.7708 
00>  AY ratio: 95.001 
00>  AZ ratio: 218.31 
00>  should be between 50 and 150% 
00>  GX ratio: 12.4669  
00>  GY ratio: 12.4005  
00>  GZ ratio: 12.4143  
00>  should be between 50 and 150%

kriswiner, 2022-08-18:
Are you setting the accel and gyro configuration properly?

_i2c_bus->writeByte(ICM42688_ADDRESS, ICM42688_REG_BANK_SEL, 0x00); //
select register bank 0
_i2c_bus->writeByte(ICM42688_ADDRESS, ICM42688_PWR_MGMT0, 0x0F); // turn on
accel and gyro in low noise mode
delay(1);
_i2c_bus->writeByte(ICM42688_ADDRESS, ICM42688_ACCEL_CONFIG0, AFS_4G << 5 |
AODR_1kHz); // FS = 2
_i2c_bus->writeByte(ICM42688_ADDRESS, ICM42688_GYRO_CONFIG0, GFS_250DPS << 5
| GODR_1kHz); // FS = 3
_i2c_bus->writeByte(ICM42688_ADDRESS, ICM42688_GYRO_ACCEL_CONFIG0, 0x44); //
set gyro and accel bandwidth to ODR/10

kriswiner, 2022-08-18:
Looks like the main code is not quite right. That is, if you select an
accel scale different from 2 G or a gyro scale different from 250dps the
self test will be wrong.

rohithr2597, 2022-08-18:
I just messed up with setting up the register gyro config0. I didn't do left shift by 5 (GFS_250DPS << 5) . 

Now it's working fine. Thank you for your suggestion and quick reply!!

rohithr2597, 2022-08-19:
> Accelerometer self-test works fine but the gyro always fails in the self-test. I tried debugging using the library and checked everything but nothing worked. 00> accelerometer self test 00> Ax dif : [...]

AZ ratio seems to be high in this. But other self-tests are passed for accel and gyro.

kriswiner, 2022-08-19:
These results are not using the correct scale factor, but for ratios
shouldn't matter. But might want to try again with correct scale factors.

In any case, self test is not critical; it has no bearing on the quality of
the data output unless the values are so far off those expected that the
sensor is likely broken.

closed 2022-08-28 by rohithr2597

kriswiner, 2022-10-11:
Fixed now.