kriswiner/ICM42688 #7: Add an open source license, such as MIT and/or Apache 2.0

TheButlah, 2023-03-07:
I'd like to reference this code in some other work, but without an open source license, I won't be able to look at it.

Could you slap a MIT and/or Apache 2.0 license on there?

kriswiner, 2023-03-07:
> Could you add a MIT and/or Apache 2.0 license?

The license for the code is as follows:

"Library may be used freely and without limit with proper attribution."

Anything you might add would be more restrictive, but you are free to do so
as long as the attribution command is followed.

TheButlah, 2023-03-12:
Personally, I'd feel more comfortable if it had a commonly recognized license instead of an ad-hoc/custom but permissive one. But thanks for the clarification at least :)

kriswiner, 2023-03-12:
You may add one, as long as you respect the original (broader albeit
ad-hoc) license.

closed 2023-03-19 by TheButlah